hbrobotics/ros_arduino_bridge #16: Problem with loading type ros_arduino_msgs/ServoWrite

vovaekb, 2015-01-11:
When I ran command for controlling servo I got an error: "Unable to load type [ros_arduino_msgs/ServoWrite]". The same error appeared for reading servo and writing to digital pin. What I missed during installation and configuration your package?

pirobot, 2015-01-13:
@vovaekb  Can you please tell us a few things about your setup first?  In particular, we need to know what ROS version you are using and also what version of Ubuntu.  Then please tell us how you installed the ros_arduino_bridge package.  Did you run catkin_make after cloning the Git repository?  Finally, can you send us the command you tried to run that generated the error above?

Thanks!
patrick

vovaekb, 2015-01-14:
I installed ros_arduino_bridge on Ubuntu 12.04, I have ROS of version Hydro. Installation was held strictly in accordance with manual in README of your package. I ran catkin_make after cloning the Git repository. My failed commands are follows:
$ rosservice call /arduino/servo_write id pos
$ rosservice call /arduino/servo_read id

pirobot, 2015-01-15:
Hmmm.  I cannot reproduce this problem and I am using the same setup as you.  When you run catkin_make, can you verify that there are no build errors in any of your packages?  I'm wondering if the build is failing somewhere before it builds the ros_arduino_msgs package.  Also, what output to you see when you run:

$ cd ~/catkin_ws
$ find . -name ServoWrite\\*

You should see something like this:

./devel/include/ros_arduino_msgs/ServoWriteResponse.h
./devel/include/ros_arduino_msgs/ServoWrite.h
./devel/include/ros_arduino_msgs/ServoWriteRequest.h
./devel/share/common-lisp/ros/ros_arduino_msgs/srv/ServoWrite.lisp

vovaekb, 2015-01-15:
There is no build errors during compiling catkin_workspace. I see almost the same output for this command except last line:
./src/ros_arduino_bridge/ros_arduino_msgs/srv/ServoWrite.srv

pirobot, 2015-01-15:
OK, I think I finally found the source of your problem.  Make sure you have the following line toward the end of your ~/.bashrc file:

source ~/catkin_ws/devel/setup.bash

Now, every new terminal will be able to see the catkin packages, messages and services that live in ~/catkin_ws/src.

vovaekb, 2015-01-16:
I added that line to ~/.bashrc and now it works. Thank you very much!

pirobot, 2015-01-16:
Great!  That tip should really be in the ROS installation/setup guide...

closed 2015-01-16 by pirobot