hbrobotics/ros_arduino_bridge #20: Publisher needs queue size

gtaylor40, 2015-07-22:
Shouldn't lines such as following line 166 of arduino_sensors.py

self.pub = rospy.Publisher("~sensor/" + self.name, Range)

be changed to include queue_size as follows 

self.pub = rospy.Publisher("~sensor/" + self.name, Range, queue_size=10)

as well as other similar files in the ros_arduino_bridge software?

Without the queue_size being defined, several errors were reported when executing:

roslaunch ros_arduino_python arduino.launch

mikeferguson, 2015-07-22:
@gtaylor40 the queue_size thing is new in Hydro and required in Indigo and later -- so I think it just is a matter of this package having not yet been updated yet. A pull request with the required fixes is probably the best way to get them updated.

gtaylor40, 2015-07-23:
Mike,

Thanks for the info. It really tickled me to have been able to figure out how to modify the code by adding queue_size=10 in several places to allow running successfully the Arduino Bridge software and displaying Ping information. My compliments to the code developer(s) whose error message not only indicated the error, but also gave a URL that provided information as to how to correct it.

George Taylor

https://github.com/hbrobotics/ros_arduino_bridge/issues/20#issuecomment-123849239

pirobot, 2015-08-25:
@gtaylor40 - the queue_size parameter has been added to the Publishers in the indigo-devel branch.

closed 2015-08-25 by pirobot