hbrobotics/ros_arduino_bridge #23: problem reading both encoders

leo100du, 2015-10-21:
I have an arduino uno on board and when i put 'e'(read encoders) after moving both motors with (m 50 50, or any numbers), i dont have the same or proximate reading, i get something like this: 2 3259, refering 2--> to the left encoder counter, and 3259 the right encoder counter, my connections are 
Left wheel encoder A output -- Arduino UNO pin 2
Left wheel encoder B output -- Arduino UNO pin 3

Right wheel encoder A output -- Arduino UNO pin A4
Right wheel encoder B output -- Arduino UNO pin A5

 i've tried to change left encoder pins but i get always similar results
wich could be the solution to this problem?

kishorekumar29, 2016-03-16:
Didi you get around this issue and manage to get perfect results?

sgzuccaro42, 2016-05-27:
Has anyone found a solution for this? I am having the same issue. After many hours of troubleshooting, I still can not read encoder counts on pin 2 and pin 3. I have tried different uno's, different encoders, looking at the encoder pwm with an oscilloscope, reading and writing to pin 2 and 3 (which works), change to other ports... Any advice would be greatly appreciated.

DerKorb, 2016-09-27:
For anyone else struggling with this, the vnh5019 shield also uses pin 2 and overwrites it as output.

roboticsai, 2017-03-09:
Well i used ros arduino package from master branch, and its working, i can read the encoder values from pin 2 and 3. as e 3 or e 2. indigo branch has got bug. master branch is working.

roboticsai, 2017-03-10:
Can someone tell me which branch of ros arduino package are people using. Cas i'm not able to read encoder values no matter which branch of package i use.