hbrobotics/ros_arduino_bridge #24: Hardware selection to implement the package.

kishorekumar29, 2016-03-11:
I am developing ROS navigation robot, after some research i have zeroed on this package for differential drive and encoder reading. I have my Arduino UNO and planning to buy the this Pololu [motor](https://www.pololu.com/product/2827) and [motor driver.](https://www.pololu.com/product/2503) 
Can i use my Arduino UNO to control motor and read quadrature encoder data(available in motor), if not what are all the boards i need to buy, only Robogaia encoder shield or also Arduino Mega.

PS: I read some posts noted that Arduino UNO and Mega cant handle the motor encoder because the they don't have enough clock speed. Required clock speed in 50MHz.

pirobot, 2016-03-11:
If you look toward the end of the README file, there is a section explaining how to use an Arduino Uno and direct encoder inputs.  Someone else added that support and I haven't tried it myself but if it works, it should not need any other hardware.

kishorekumar29, 2016-03-14:
Is there any chance that i can use[ this ](https://www.pololu.com/product/708 )motor driver for the code

pirobot, 2016-03-15:
I don't know for sure.  The supported Pololu motor drivers (VNH5019 and MC33926) are actually Arduino shields that come with an Arduino library.  We use those libraries and header files in the ros_arduino_bridge firmware to communicate with the motor drivers.

closed 2016-05-27 by pirobot