hbrobotics/ros_arduino_bridge #26: Error while compiling ROSArduinoBridge

kishorekumar29, 2016-04-08:
I followed the given steps and installed the sketch files to the sketchbook folder. I also installed the dependent sketches of motor and encoder but when i tried to compile the code i get the below error

```
> ROSArduinoBridge.cpp.o: In function `loop':
/usr/share/arduino/ROSArduinoBridge.ino:280: undefined reference to `MegaEncoderCounter::YAxisGetCount()'
/usr/share/arduino/ROSArduinoBridge.ino:325: undefined reference to `MegaEncoderCounter::XAxisGetCount()'
/usr/share/arduino/ROSArduinoBridge.ino:335: undefined reference to `MegaEncoderCounter::YAxisReset()'
/usr/share/arduino/ROSArduinoBridge.ino:335: undefined reference to `MegaEncoderCounter::XAxisReset()'
/usr/share/arduino/ROSArduinoBridge.ino:335: undefined reference to `DualVNH5019MotorShield::init()'
/usr/share/arduino/ROSArduinoBridge.ino:335: undefined reference to `DualVNH5019MotorShield::setM1Speed(int)'
/usr/share/arduino/ROSArduinoBridge.ino:335: undefined reference to `DualVNH5019MotorShield::setM2Speed(int)'
/usr/share/arduino/ROSArduinoBridge.ino:335: undefined reference to `MegaEncoderCounter::XAxisGetCount()'
/usr/share/arduino/ROSArduinoBridge.ino:335: undefined reference to `MegaEncoderCounter::XAxisGetCount()'
/usr/share/arduino/ROSArduinoBridge.ino:335: undefined reference to `MegaEncoderCounter::MegaEncoderCounter(unsigned char)'
/usr/share/arduino/ROSArduinoBridge.ino:335: undefined reference to `DualVNH5019MotorShield::DualVNH5019MotorShield()'
collect2: error: ld returned 1 exit status

```

pirobot, 2016-04-08:
@kishorekumar29 - Did you install the motor and encoder libraries in the _libraries_ subfolder of your sketchbook folder?  They must go in the libraries subfolder or the Arduino IDE will not pick them up.  On the other hand, the ROSArduinoBridge folder just goes in the top-level sketchbook folder.

kishorekumar29, 2016-04-15:
Thanks for the direction, i installed the libraries as per your advice and resolved that error but now i bumped into new error as below

```
`/home/robotics/sketchbook/libraries/MegaEncoderCounter/MegaEncoderCounter.cpp: In constructor ‘MegaEncoderCounter::MegaEncoderCounter(unsigned char)’:
/home/robotics/sketchbook/libraries/MegaEncoderCounter/MegaEncoderCounter.cpp:13:4: error: ‘DDRA’ was not declared in this scope
    DDRA = B00000000;   // sets Arduino Mega (ATMEL ATMEGA) Digital pins 22(PORTA0) to 29(PORTA7) as inputs from HCTL-2032 - D0 to D7 
    ^

/home/robotics/sketchbook/libraries/MegaEncoderCounter/MegaEncoderCounter.cpp: In member function ‘long unsigned int MegaEncoderCounter::XAxisGetCount()’:
/home/robotics/sketchbook/libraries/MegaEncoderCounter/MegaEncoderCounter.cpp:65:14: error: ‘PINA’ was not declared in this scope
    busByte = PINA;
              ^

/home/robotics/sketchbook/libraries/MegaEncoderCounter/MegaEncoderCounter.cpp: In member function ‘long unsigned int MegaEncoderCounter::YAxisGetCount()’:
/home/robotics/sketchbook/libraries/MegaEncoderCounter/MegaEncoderCounter.cpp:113:14: error: ‘PINA’ was not declared in this scope
    busByte = PINA;
```

              ^`

pirobot, 2016-04-16:
Have you set the Arduino Board type in the IDE to match the actual Arduino board you are using?  (E.g. Mega versus Uno etc.)  Also, are you using the original [2-axis encoder](http://www.robogaia.com/two-axis-encoder-counter-mega-shield-version-2.html) shield or the newer [3-axis encoder](http://www.robogaia.com/3-axis-encoder-conter-arduino-shield.html) shield?

kishorekumar29, 2016-04-16:
I am not using any board because i am trying to load the code into board, i am just trying to compile. The error occurs while compiling the code.

pirobot, 2016-04-17:
You still need to select the correct board in the Arduino IDE because otherwise certain pin definitions will not exist.  So select the board "Arduino/Genuino Mega or Mega 2560" and that compile error should go away.

closed 2016-05-27 by pirobot