hbrobotics/ros_arduino_bridge #27: roslaunch ERROR

Sunlcy, 2016-05-26:
When I use `roslaunch ros_arduino_python arduino.launch`, I get this: ERROR: cannot launch node of type [ros_arduino_python/arduino_node.py]: can't locate node [arduino_node.py] in package [ros_arduino_python].

pirobot, 2016-05-26:
Can you "roscd ros_arduino_bridge" without error?

Sunlcy, 2016-05-27:
Yes, it run without any error. And I met another problem(issue #28). @pirobot

pirobot, 2016-05-27:
It sounds like the arduino_node.py file is not executable for you which is odd.  Try the following commands:

   roscd ros_arduino_python/nodes
   chmod u+x arduino_node.py
   roslaunch ros_arduino_python arduino.launch

Sunlcy, 2016-05-28:
I tried these commands, but it output:
> Connecting to Arduino on port /dev/ttyACM0 ...
> Failed to determine baud rate of Arduino so aborting!
> ...

pirobot, 2016-05-28:
Is your Arduino really connected to /dev/ttyACM0?  Make sure you have no other USB devices connected and try again.

Also, try increasing the timeout value in your arduino_params.yaml file.  I think the default is 0.1 seconds which is a little too short.  Bump it up to 0.5 or even 1.0 and this will give the driver more time to wake up the serial port.

Sunlcy, 2016-05-28:
Yes, I'm sure my Arduino is connected to /dev/ttyACM0 and no other USB devices. Besides, I increase the timeout value, but it still doesn't work. I looked through source codes and debugged them, and I think the problem may come from the function `recv()` in file "arduino_driver.py", because it can't return correct value from serial, every time the program execute command.

pirobot, 2016-05-28:
Can you please give me some more info about your setup?  (1) What version of ROS are you using,  (2) what branch of the ros_arduino_bridge package are you using (e.g. hydro-devel, indigo-devel) , (3) what type of Arduino are you using and (4) what kind of computer are you running ROS on (e.g. laptop versus Raspberry Pi etc).

Sunlcy, 2016-05-28:
Info you want to know is as follows:

1. The version of ROS I'm using is indigo;
1. The branch is master;
1. My arduino is arduino Mega 2560;
1. My computer is laptop only.

pirobot, 2016-05-28:
Please try the indigo-devel branch since the master branch is under active development and is not guaranteed to work.

pirobot, 2016-05-29:
@Sunlcy Does that mean things are working for you now with the indigo-devel branch?

Sunlcy, 2016-05-29:
@pirobot Very sorry replying to you so late! I need some time to make the indigo-devel branch compatible with my hardware, and I am a little busy recently. I'm trying to spare time to do it, and I will tell you the result  as soon as I finish it.

Sunlcy, 2016-06-01:
@pirobot I tested the indigo-devel branch on my computer and arduino board, it works fine. So I think there must be some bugs about the baudrate in the master branch. Anywhere, thanks your help very much! :smile:

pirobot, 2016-06-01:
That's good to hear!  I have put a note on top of the master branch README file warning people not to use it.  I am adding a number of new features to master and testing them out before merging them to the indigo-devel branch later on.

closed 2016-06-01 by pirobot

yangqiao95, 2019-09-05:
Traceback (most recent call last):
  File "/home/yangqiao/catkin_ws/src/bsbot/bsbot_python/nodes/arduino_node.py", line 24, in <module>
    from bsbot_python.arduino_driver import Arduino
  File "/home/yangqiao/catkin_ws/src/bsbot/bsbot_python/src/bsbot_python/arduino_driver.py", line 264
    def get_baud(self):
                      ^
IndentationError: unindent does not match any outer indentation level
[arduino-2] process has died [pid 8072, exit code 1, cmd /home/yangqiao/catkin_ws/src/bsbot/bsbot_python/nodes/arduino_node.py __name:=arduino __log:=/home/yangqiao/.ros/log/c8dfef38-cfba-11e9-8675-50e5498171c8/arduino-2.log].
log file: /home/yangqiao/.ros/log/c8dfef38-cfba-11e9-8675-50e5498171c8/arduino-2*.log

yangqiao95, 2019-09-05:
问题，咋改Traceback (most recent call last):
  File "/home/yangqiao/catkin_ws/src/bsbot/bsbot_python/nodes/arduino_node.py", line 24, in <module>
    from bsbot_python.arduino_driver import Arduino
  File "/home/yangqiao/catkin_ws/src/bsbot/bsbot_python/src/bsbot_python/arduino_driver.py", line 264
    def get_baud(self):
                      ^
IndentationError: unindent does not match any outer indentation level
[arduino-2] process has died [pid 8072, exit code 1, cmd /home/yangqiao/catkin_ws/src/bsbot/bsbot_python/nodes/arduino_node.py __name:=arduino __log:=/home/yangqiao/.ros/log/c8dfef38-cfba-11e9-8675-50e5498171c8/arduino-2.log].
log file: /home/yangqiao/.ros/log/c8dfef38-cfba-11e9-8675-50e5498171c8/arduino-2*.log