hbrobotics/ros_arduino_bridge #29: Can not find what encoder counter to get

sgzuccaro42, 2016-05-27:
After many hours of troubleshooting, I still can not read encoder counts on pin 2 and pin 3. I have tried different uno's, different encoders, looking at the encoder pwm with an oscilloscope, reading and writing to pin 2 and 3 (which works), change to other ports... Any advice would be greatly appreciated.

I decided I may break down and use a mega with an encoder counter. Since the robogaia is discontinued, does anyone have advice on what other encoder counter works with the recommended motor drivers?

AnasIbrahim, 2016-05-27:
are you sure you:
1- uncomment the #define USE_BASE and comment out #undef USE_BASE
2- commented #define ROBOGAIA and uncommented #define ARDUINO_ENC_COUNTER
3- same for motor controller
you can use one encoder driver and one motor driver

this is how code should look: (i didn't choose a motor driver here, you have to choose one of course)
```
#define USE_BASE      // Enable the base controller code
//#undef USE_BASE     // Disable the base controller code

/* Define the motor controller and encoder library you are using */
#ifdef USE_BASE
   /* The Pololu VNH5019 dual motor driver shield */
   //#define POLOLU_VNH5019

   /* The Pololu MC33926 dual motor driver shield */
   //#define POLOLU_MC33926

   /* The RoboGaia encoder shield */
   //#define ROBOGAIA
   
   /* Encoders directly attached to Arduino board */
   #define ARDUINO_ENC_COUNTER
#endif
```
you will find these instructions on the Note section here on github

sgzuccaro42, 2016-05-27:
Thank you for your quick response. This is my first time reaching out for help, it is nice to talk to someone about this :) I believe i have done what you described. I attached what i have below. I am only moderately familiar with using arduino. any chance i'm missing some type of interrupt initialization? Im shooting in the dark now, do you have any more suggestions? Any troubleshooting approaches i could take?

`#define USE_BASE      // Enable the base controller code
//#undef USE_BASE     // Disable the base controller code

/* Define the motor controller and encoder library you are using */
#ifdef USE_BASE
   /* The Pololu VNH5019 dual motor driver shield */
   #define POLOLU_VNH5019

   /* The Pololu MC33926 dual motor driver shield */
   //#define POLOLU_MC33926

   /* The RoboGaia encoder shield */
   //#define ROBOGAIA
   
   /* Encoders directly attached to Arduino board */
   #define ARDUINO_ENC_COUNTER
#endif`

AnasIbrahim, 2016-05-27:
1- the only thing i can notice here is that you mentioned pin 2 and 3 which are used for the same encoder but you didn't mention trying A4 & A5.
what type of encoders are you trying to connect? is magnetic? (it suppose to be)
2- also what changes you made for the pin the code mentioned a note about PORTD & PORTC
```
#ifdef ARDUINO_ENC_COUNTER
  //below can be changed, but should be PORTD pins; 
  //otherwise additional changes in the code are required
  #define LEFT_ENC_PIN_A PD2  //pin 2
  #define LEFT_ENC_PIN_B PD3  //pin 3
  
  //below can be changed, but should be PORTC pins
  #define RIGHT_ENC_PIN_A PC4  //pin A4
  #define RIGHT_ENC_PIN_B PC5   //pin A5
#endif
```

sgzuccaro42, 2016-05-27:
Oh, it is an 500 cpr optical encoder on a Pittman motor. So doing quadrature encoding makes that's 2000 cpr. Is there a way to adjust for this not being a magnetic encoder? As far as the other comments, I have used A4 and A5 and that actually works with both wheel encoders while nether work on PD2 and PD3. That would seem to elimate it being the encoder but perhaps there is a timing/speed difference between portC and portD or some ther difference.

I have gone back to the default ports but I mentioned I tried other ports such as PD4 and PD5 which was no better.

AnasIbrahim, 2016-05-27:
if that encoder has the same interface it's not a problem, i was just mentioning the usual case, and as you mentioned if it worked with A4 & A5 it can't be the problem.
i have no more ideas :D

sgzuccaro42, 2016-05-27:
I was thinking, could I change the interrupt to only read rising edges? Because I don't need quadrature resolution; 500 car is plenty. Using quadrature encoding, with my gearing and wheel diameter I am getting about 39,000 encoder counts per wheel rotation.

sgzuccaro42, 2016-05-27:
I appreciate your effort and time. Thank you so much!

KristofRobot, 2016-05-28:
@sgzuccaro42 - some additional questions to better understand the problem:

(1) You state you cannot read encoder counts on pins 2-3 - do you mean that encoder counts stay 0? Or that they work incorrectly?

(2) Could you share the oscilloscope traces you made?

(3) Could you make the exact code you are using available somewhere (e.g. gist) so it can be reviewed?

(4) Which release are you using? hydro-release? master? indigo-release?

sgzuccaro42, 2016-05-29:
Thank you for picking this up KristofRobot. I will do my best to help you help me. :)

(1) the counts stay "0 0". Actually, it typically starts a "-1 0" then as soon as barely rotate the wheel it becomes and stays "0 0". Ocassonially, it'll bump to "1 0" or even "2 0" and then go back to "0 0", but I can't get anything repeatable and it is a rare thing.

(2) itll be a week till I'm back at school, but for now I can say it was a clean square wave with expected amplitude and frequency. 

(3) I will do this. And will post when I get it. I haven't ever before so I'll look in to how to do it.

(4) I am using indigo on Linux mint. And the latest arduino ide.

sgzuccaro42, 2016-05-29:
Alright, i made a glist. this is my first time so bare with me and please let me know if you cant find it or something. it is titled:
sgzuccaro42/ROSArduinoBridge

and the link is:
https://gist.github.com/sgzuccaro42/6c3c2be5de85ab478abb82c457754c5a

KristofRobot, 2016-05-29:
@sgzuccaro42 - if I use your code with my MC33926 dual motor driver shield it works perfectly.  

So it seems there is a conflict between the VNH5019 dual motor driver shield and the Arduino encoder implementation. Indeed, it turns out that that motor shield is using pin 2 - see https://www.pololu.com/docs/0J49/3.c.

You'll have to either change the motor shield, or the arduino encoder code - the latter is probably easiest. To do so, edit  ``encoder_driver.h`` and replace pin 2 (PD2) by another PORTD pin. I would stay away from PD0/PD1 (Serial), that only leaves pin 5 (PD5).

KristofRobot, 2016-05-29:
You also need to change the code in ``encoder_driver.ino`` to reflect this change (assuming you are using pin 5 and pin 3):

Replace:
```
enc_last |= (PIND & (3 << 2)) >> 2; //read the current state into lowest 2 bits
```

by

```
enc_last |= (PIND & (5 << 3)) >> 2; //read the current state into lowest 2 bits
```

I just tested this with my MC33926 driver shield, and that works.

sgzuccaro42, 2016-05-29:
KristofRobot, I could not be more thankful for this help! It'll be a week before I'm back to test it, but I'm sure this will fix it! I will repost when I test so we can close this. Thank you again.

sgzuccaro42, 2016-06-06:
I am finally back and testing this code. I have made all the suggested changes, and see an improvement! :) It still not working correctly, but it seems that we have some helpful clues. 

The encoder on PD5 and PD3 will read half the value of the other encoder (which reads correctly, and both encoders are rotating at the same speed). Further, this is only the case in the forward (positive) direction. when the encoders are rotated backward (as in pushing the robot backward) the encoder on PD5 and PD3 does not recognize the change - the value remains whatever it was before the move.

KristofRobot, 2016-06-08:
Yes, sorry, I see now that this cannot work (it seemed to work during a very quick test, but I only tested one direction, and rather quickly).

The problem is the code enc_last expects to find the info in the lowest two bits, while the command above in fact puts the info in the lowest second and fourth bit - causing this to work partially. This complication is caused by the fact that we are using two ports that are not next to each other. 

Try the following instead:

```
enc_last |= (((PIND & (1 << 3)) >> 2) | ((PIND & (1 << 5)) >> 5)); //read the current state into lowest 2 bits
```
I tested this a bit more extensively on the bench myself, and that seems to work fine.

KristofRobot, 2016-06-08:
I should also add that the above is with the following pin assignments:

```
   #define LEFT_ENC_PIN_A PD5  //pin 5
   #define LEFT_ENC_PIN_B PD3  //pin 3
```

sgzuccaro42, 2016-06-09:
This solution worked wonderfully! Thank you very much for your extensive help. I imagine this solution will also help others attempting to use this driver, it has certainly saved me. :)

closed 2016-06-09 by sgzuccaro42

Kaustav6422, 2017-10-25:
Thank you so much for this solution. I have been struggling with the same problem.