hbrobotics/ros_arduino_bridge #30: Does this package support non-pololu drivers?

kishorekumar29, 2016-09-03:
I am using my own motor driver instead of pololu drivers.
Do this package support the drivers other than which it is developed for., in that case what are all the changes need to be done?

RobotCdf, 2016-09-04:
I'm sorry, but I am not the original author of this project. I'm also doing
this direction, using DFrobot drive, also has the same problem, which is in
testing now.

pirobot, 2016-09-05:
@kishorekumar29 - Yes, you can certainly add support for other motor drivers if you know something about Arduino programming.  Most motor drivers come with an Arduino library so first you need to install that.  Then take a look at the motor_driver.ino and motor_driver.h files under ros_arduino_firmware/src/libraries/ROSArduinoBridge and you can see how the current drivers are supported.  Once you add your support in those two files, you'll need to add a corresponding #define in the main file ros_arduino_firmware/src/libraries/ROSArduinoBridge/ROSArduinoBridge.ino.

patrickpoirier51, 2017-08-19:
Just finished building a diff-drive using cheap dual L298N.
I simply remapped the pins of the 
DualVNH5019MotorShield::DualVNH5019MotorShield()
{
  //Pin map
  _INA1 = 2;
  _INB1 = 3;
  _INA2 = 7;
  _INB2 = 8;
}

and changed timer ICR1 register value from 400 to 300

Works OK,  but obviously there is no current feedback