hbrobotics/ros_arduino_bridge #31: Use Ctrl+C on an Embedded Board can cause re-connection problem

sonictl, 2016-11-11:
I cannot connect the Arduino board after I used Ctrl+C to terminate the ros_arduino_python/arduino.launch
I get: 

> Connecting to Arduino on port /dev/ARDUINOBASE ...
Serial Exception:
(<class 'serial.serialutil.SerialException'>, SerialException(), <traceback object at 0xb5f57a80>)
Traceback follows:
Traceback (most recent call last):
File "/root/catkin_ws/src/ros_arduino_bridge/ros_arduino_python/src/ros_arduino_python/arduino_driver.py", line 73, in connect
raise SerialException
SerialException
Cannot connect to Arduino!

Hardware List:
  UP-level computer: OrangePi running lubuntu 14.04
  Low-level board: Arduino Mega 2560 with CH340(usb-serial converter)

pirobot, 2016-11-12:
Do you have the same issue when using the master branch of the ros_arduino_bridge package?  I ask because I have made many changes to that branch to improve the serial connection and I have not yet merged those changes with the indigo-devel branch.

sonictl, 2016-11-16:
I do have the same issue when using master branch. maybe you need to have a Raspberry-like board and have a test. THX!

pirobot, 2016-11-16:
Unfortunately I don't have a Raspberry-like board.  You might try posting your issue to the [Home Brew Robotics forum](http://groups.google.com/group/hbrobotics) since many people in the club use RPi platforms with their robots.

sonictl, 2016-11-17:
I noted that I'm also using an other serial_based hardware ROS driver which is for neato_xv11 LiDAR. It is written by C++ and works well. Maybe we can try to write a C++ version for ros_arduino_bridge. How do you think? ;)

pirobot, 2016-11-17:
Yeah, I noticed the same good behavior with the Arduino IDE a while back which is also written in C++.  I just double-checked the arduino_python.py node in the master branch and for some unknown reason, I had commented out the line that closes the serial port on shutdown.  I just un-commented that line and pushed the update so please do a git pull and see if that fixes the problem.

sonictl, 2016-11-21:
-----------------------
I met the errors below when I was trying to launch the ros_arduino_python/launch/arduino.launch file.
-----------------------
>  new master
process[master]: started with pid [2677]
ROS_MASTER_URI=http://pirobot:11311
setting /run_id to 7f6362c4-afea-11e6-b06c-28f3667b4dc0
process[rosout-1]: started with pid [2690]
started core service [/rosout]
process[arduino-2]: started with pid [2694]
Traceback (most recent call last):
  File "/root/catkin_ws/src/ros_arduino_bridge/ros_arduino_python/nodes/arduino_node.py", line 24, in <module>
    from ros_arduino_python.arduino_sensors import *
  File "/root/catkin_ws/src/ros_arduino_bridge/ros_arduino_python/src/ros_arduino_python/arduino_sensors.py", line 26, in <module>
    from ros_arduino_python.diagnostics import DiagnosticsUpdater
  File "/root/catkin_ws/src/ros_arduino_bridge/ros_arduino_python/src/ros_arduino_python/diagnostics.py", line 23, in <module>
    import diagnostic_updater
ImportError: No module named diagnostic_updater
[arduino-2] process has died [pid 2694, exit code 1, cmd /root/catkin_ws/src/ros_arduino_bridge/ros_arduino_python/nodes/arduino_node.py __name:=arduino __log:=/root/.ros/log/7f6362c4-afea-11e6-b06c-28f3667b4dc0/arduino-2.log].
log file: /root/.ros/log/7f6362c4-afea-11e6-b06c-28f3667b4dc0/arduino-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

-----------------------
After I git clone the diagnostics package and catkin_make this package, I met the errors below when I was trying to launch the ros_arduino_python/launch/arduino.launch file.
-----------------------

>NODES
  /
    arduino (ros_arduino_python/arduino_node.py)
auto-starting new master
process[master]: started with pid [3034]
ROS_MASTER_URI=http://pirobot:11311
setting /run_id to 256dd7de-afec-11e6-b06c-28f3667b4dc0
process[rosout-1]: started with pid [3047]
started core service [/rosout]
process[arduino-2]: started with pid [3063]
Traceback (most recent call last):
  File "/root/catkin_ws/src/ros_arduino_bridge/ros_arduino_python/nodes/arduino_node.py", line 30, in <module>
    from ros_arduino_python.follow_controller import FollowController
  File "/root/catkin_ws/src/ros_arduino_bridge/ros_arduino_python/src/ros_arduino_python/follow_controller.py", line 32, in <module>
    from control_msgs.msg import FollowJointTrajectoryAction
ImportError: No module named control_msgs.msg
[arduino-2] process has died [pid 3063, exit code 1, cmd /root/catkin_ws/src/ros_arduino_bridge/ros_arduino_python/nodes/arduino_node.py __name:=arduino __log:=/root/.ros/log/256dd7de-afec-11e6-b06c-28f3667b4dc0/arduino-2.log].
log file: /root/.ros/log/256dd7de-afec-11e6-b06c-28f3667b4dc0/arduino-2*.log

pirobot, 2016-11-21:
Please install the ROS diagnostic_updater and control_msgs packages as follows:

`sudo apt-get install ros-indigo-diagnostic-updater ros-indigo-control-msgs`

sonictl, 2016-11-22:
It seems solved. Thank you very much and hope you can fix the indigo-devel soon.

sonictl, 2016-11-22:
The reconnection probelm is solved when I use the master branch. Thank you very much!!!
I guess you may need to fix the indigo-devel branch in case the master branch is unstable and may cause other problems.

sonictl, 2016-11-29:
> root@piRobot:~/catkin_ws# roslaunch ros_arduino_python arduino.launch 
>  logging to /root/.ros/log/43f146e6-b5e6-11e6-b06c-28f3667b4dc0/roslaunch-piRobot-6546.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://piRobot:42719/
> 
> SUMMARY
> = = = = = = = =
> 
> PARAMETERS
>  * /arduino/Kd: 12
>  * /arduino/Ki: 0
>  * /arduino/Ko: 50
>  * /arduino/Kp: 10
>  * /arduino/accel_limit: 1.0
>  * /arduino/base_controller_rate: 10
>  * /arduino/base_frame: base_link
>  * /arduino/baud: 57600
>  * /arduino/encoder_resolution: 2200
>  * /arduino/gear_reduction: 1.0
>  * /arduino/motors_reversed: False
>  * /arduino/port: /dev/ARDUINOBASE
>  * /arduino/rate: 50
>  * /arduino/sensors/arduino_led/direction: output
>  * /arduino/sensors/arduino_led/pin: 13
>  * /arduino/sensors/arduino_led/rate: 5
>  * /arduino/sensors/arduino_led/type: Digital
>  * /arduino/sensorstate_rate: 10
>  * /arduino/timeout: 0.1
>  * /arduino/use_base_controller: True
>  * /arduino/wheel_diameter: 0.078
>  * /arduino/wheel_track: 0.36
>  * /rosdistro: indigo
>  * /rosversion: 1.11.20
> 
> NODES
>   /
>     arduino (ros_arduino_python/arduino_node.py)
> 
> auto-starting new master
> process[master]: started with pid [6558]
> ROS_MASTER_URI=http://piRobot:11311

> setting /run_id to 43f146e6-b5e6-11e6-b06c-28f3667b4dc0
> process[rosout-1]: started with pid [6571]
> started core service [/rosout]
> process[arduino-2]: started with pid [6583]
> Connecting to Arduino on port /dev/ARDUINOBASE ...
> Serial Exception:
> (<class 'serial.serialutil.SerialException'>, SerialException(), <traceback object at 0xb5f309b8>)
> Traceback follows:
> Traceback (most recent call last):
>   File "/root/catkin_ws/src/ros_arduino_bridge/ros_arduino_python/src/ros_arduino_python/arduino_driver.py", line 73, in connect
>     raise SerialException
> SerialException
> Cannot connect to Arduino!
> [arduino-2] process has died [pid 6583, exit code 1, cmd /root/catkin_ws/src/ros_arduino_bridge/ros_arduino_python/nodes/arduino_node.py __name:=arduino __log:=/root/.ros/log/43f146e6-b5e6-11e6-b06c-28f3667b4dc0/arduino-2.log].
> log file: /root/.ros/log/43f146e6-b5e6-11e6-b06c-28f3667b4dc0/arduino-2*.log

pirobot, 2016-11-30:
Thanks for checking.  Would you mind running the master branch for awhile?  Apparently that branch fixes the serial port problem for your RPi and having someone use the branch more regularly would be quite helpful.  Once I know the branch doesn't cause you any other problems, I'll push the changes to indigo-dev.

sonictl, 2016-12-01:
OK, I'll use that. Thx!

pirobot, 2016-12-01:
Good idea.  I have added the pre-requisites under the System Requirements section.

closed 2017-03-10 by pirobot