hbrobotics/ros_arduino_bridge #39: ros_arduino_python node died

coderchaser, 2017-06-02:
## Version: *Indigo* on RaspberryPi
I use `ros_arduino_python` to controll arduino on rpi.Lastday it worked fine.But today when I `roslaunch` the launch file,some erros happen.
```
Connecting to Arduino on port /dev/ttyUSB0 ...
Traceback (most recent call last):
  File "/opt/ros/indigo/share/ros_arduino_python/nodes/arduino_node.py", line 195, in <module>
    myArduino = ArduinoROS()
  File "/opt/ros/indigo/share/ros_arduino_python/nodes/arduino_node.py", line 87, in __init__
    self.controller.connect()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/ros_arduino_python/arduino_driver.py", line 68, in connect
    test = self.get_baud()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/ros_arduino_python/arduino_driver.py", line 261, in get_baud
    return int(self.execute('b'));
  File "/opt/ros/indigo/lib/python2.7/dist-packages/ros_arduino_python/arduino_driver.py", line 178, in execute
    return int(value)
TypeError: int() argument must be a string or a number, not 'NoneType'
```
I check the `arduino_driver.py` and I didn't find the reason.Rebooting node doesn't work.
Here is the log files:
[e86d038a-4755-11e7-8caf-b827eb1042b5.zip](https://github.com/hbrobotics/ros_arduino_bridge/files/1046783/e86d038a-4755-11e7-8caf-b827eb1042b5.zip)

coderchaser, 2017-06-02:
And sometimes the error may become another one :
```
Connecting to Arduino on port /dev/ttyUSB0 ...
Traceback (most recent call last):
  File "/opt/ros/indigo/share/ros_arduino_python/nodes/arduino_node.py", line 195, in <module>
    myArduino = ArduinoROS()
  File "/opt/ros/indigo/share/ros_arduino_python/nodes/arduino_node.py", line 87, in __init__
    self.controller.connect()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/ros_arduino_python/arduino_driver.py", line 68, in connect
    test = self.get_baud()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/ros_arduino_python/arduino_driver.py", line 261, in get_baud
    return int(self.execute('b'));
  File "/opt/ros/indigo/lib/python2.7/dist-packages/ros_arduino_python/arduino_driver.py", line 178, in execute
    return int(value)
ValueError: invalid literal for int() with base 10: ''
```

coderchaser, 2017-06-02:
@pirobot

pirobot, 2017-06-05:
Hi @coderchaser,

Can you please try the master branch and see if the problem goes away?

closed 2023-02-06 by foreverlms