hbrobotics/ros_arduino_bridge #47: Issues with encoder on pins 2 and 3

dakoner, 2017-12-06:
Hi,
I'm using ros_arduino_bridge with the Pololu 19:1 CPR motor encoders: https://www.pololu.com/product/2822
The motors and encoders are attached to an Arduino Nano or UNO.

I am consistently seeing the encoder counts on pins 2/3 downscaled by a factor of 40 (for example, while rotating the wheel 1 full revolution, the wheel on A4/A5 reports the right value and the .  I read up on the other issues (https://github.com/hbrobotics/ros_arduino_bridge/issues/29), and noticed that some motor boards use pins 2/3 and so there is a way to make the ISR use another pair of pins.  However, I am just using an unshielded Nano or UNO.  Either way, I also tried pins 3/5 with the ISR code and the #defines changed as suggested in issue 29- when I do that, I get no readings at all.

To control for variables, I've tried swapping the motors (problem remains on pins 2/3).  I've inspected the encoder wave forms, everything looks correct.

Are there other options, such as using PORTB, or mechanisms to debug why the (apparently correct) encoder pin changes aren't leading ot the right encoder counts?

dakoner, 2017-12-07:
BTW, I was able to fix this by disabling servos.  I guess somehow the servo code interferes with interrupts on pins 2 and 3.

dakoner, 2017-12-07:
Looking at servos.ino, it's clear the problem is that the servos want pins 3 and 4, and attach a servo instance to those pins.  That overlaps pins 2/3 being used for the encoder.

isokar, 2018-10-11:
Hi,
just got the same problem. by default servo is attached to the same pin as encoder(kind of dumb).

arminf82, 2019-08-27:
Sitting here two day and thinking my board library did not work! Thanks for the tip!

Femi-Tofade, 2020-04-06:
@arminf82 did the ros_arduino_bridge work for you? cause it isnt working for me