hbrobotics/ros_arduino_bridge #48: What type of L298 board are you using?

dakoner, 2017-12-11:
I spent part of the weekend wondering why the ros_arduino_firmware wasn't doing speed control on my motors.  I noticed that the code for the L298 looks like this:
https://github.com/hbrobotics/ros_arduino_bridge/blob/indigo-devel/ros_arduino_firmware/src/libraries/ROSArduinoBridge/motor_driver.ino#L76

Specifically, you're PWMing the motor direction pins:
    if (i == LEFT) { 
      if      (reverse == 0) { analogWrite(RIGHT_MOTOR_FORWARD, spd); analogWrite(RIGHT_MOTOR_BACKWARD, 0); }
      else if (reverse == 1) { analogWrite(RIGHT_MOTOR_BACKWARD, spd); analogWrite(RIGHT_MOTOR_FORWARD, 0); }
    }
    else /*if (i == RIGHT) //no need for condition*/ {
      if      (reverse == 0) { analogWrite(LEFT_MOTOR_FORWARD, spd); analogWrite(LEFT_MOTOR_BACKWARD, 0); }
      else if (reverse == 1) { analogWrite(LEFT_MOTOR_BACKWARD, spd); analogWrite(LEFT_MOTOR_FORWARD, 0); }
    }
  

However, all the L298's that I have, for example this:
https://www.amazon.com/DROK-Controller-H-Bridge-Mega2560-Duemilanove/dp/B00CAG6GX2
the speed control is the enable pin, not the FORWARD/BACKWARD pins.

So I am curious what board you have.  Otherwise, I think this code must be in error- you're supposed to PWM the enable pins, not the motor direction pins.

YikeStone, 2018-01-10:
The code is correct. to use this code you need 4 pwm pins not 2. some corrections are needed in like- LEFT condition is checked and RIGHT motor is manipulated and vice versa.

MSohan14, 2024-07-09:
Ik that it has been a long time but can you tell if you got it to work and if so how?