hbrobotics/ros_arduino_bridge #52: Tried to add Kangaroo x2 to the Ros arduino bridge

chrissunny94, 2018-03-08:
In file included from /home/vanoramonster/catkin_ws/src/ros_arduino_bridge/ros_arduino_firmware/src/libraries/ROSArduinoBridge/ROSArduinoBridge.ino:88:0:
sketch/commands.h:14:0: warning: "PING" redefined
 #define PING           'p'
 ^
In file included from /home/vanoramonster/arduino-1.8.5/hardware/tools/avr/avr/include/avr/iom2560.h:38:0,
                 from /home/vanoramonster/arduino-1.8.5/hardware/tools/avr/avr/include/avr/io.h:174,
                 from /home/vanoramonster/arduino-1.8.5/hardware/tools/avr/avr/include/avr/pgmspace.h:90,
                 from /home/vanoramonster/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:28,
                 from /home/vanoramonster/catkin_ws/src/ros_arduino_bridge/ros_arduino_firmware/src/libraries/ROSArduinoBridge/ROSArduinoBridge.ino:82:
/home/vanoramonster/arduino-1.8.5/hardware/tools/avr/avr/include/avr/iomxx0_1.h:239:0: note: this is the location of the previous definition
 #define PING    _SFR_IO8(0x12)

 ^
/home/vanoramonster/catkin_ws/src/ros_arduino_bridge/ros_arduino_firmware/src/libraries/ROSArduinoBridge/ROSArduinoBridge.ino: In function 'void resetCommand()':
/home/vanoramonster/catkin_ws/src/ros_arduino_bridge/ros_arduino_firmware/src/libraries/ROSArduinoBridge/ROSArduinoBridge.ino:146:7: warning: converting to non-pointer type 'char' from NULL [-Wconversion-null]
   cmd = NULL;
       ^
/home/vanoramonster/catkin_ws/src/ros_arduino_bridge/ros_arduino_firmware/src/libraries/ROSArduinoBridge/ROSArduinoBridge.ino: In function 'void loop()':
/home/vanoramonster/catkin_ws/src/ros_arduino_bridge/ros_arduino_firmware/src/libraries/ROSArduinoBridge/ROSArduinoBridge.ino:298:34: warning: converting to non-pointer type 'char' from NULL [-Wconversion-null]
       if (arg == 1) argv1[index] = NULL;
                                  ^
`/home/vanoramonster/catkin_ws/src/ros_arduino_bridge/ros_arduino_firmware/src/libraries/ROSArduinoBridge/ROSArduinoBridge.ino:299:39:` warning: converting to non-pointer type 'char' from NULL [-Wconversion-null]
       else if (arg == 2) argv2[index] = NULL;
                                       ^
/home/vanoramonster/catkin_ws/src/ros_arduino_bridge/ros_arduino_firmware/src/libraries/ROSArduinoBridge/ROSArduinoBridge.ino:308:22: warning: converting to non-pointer type 'char' from NULL [-Wconversion-null]
         argv1[index] = NULL;
                      ^
libraries/SoftwareSerial/SoftwareSerial.cpp.o (symbol from plugin): In function `SoftwareSerial::read()':
(.text+0x0): multiple definition of `__vector_11'
sketch/ROSArduinoBridge.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
/home/vanoramonster/arduino-1.8.5/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld: Disabling relaxation: it will not work with multiple definitions
libraries/SoftwareSerial/SoftwareSerial.cpp.o (symbol from plugin): In function `SoftwareSerial::read()':
(.text+0x0): multiple definition of `__vector_10'
sketch/ROSArduinoBridge.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

chrissunny94, 2018-03-08:
The error is with softwareSerial .

I cant include the same , where could be the problem ?

Femi-Tofade, 2020-03-03:
How did you solve this?

chrissunny94, 2020-04-30:
@Femi-Tofade 

https://github.com/chrissunny94/kangaroo_x2_ros_arduino

This is the code i worked on for achieving this . 

I am sorry , its not documented .  Been a long time since i used this and dont remember much . 

Hope you find this helpful