hbrobotics/ros_arduino_bridge #66: Encoder reading using Ardunio Mega

Mubashir-alam, 2024-06-02:
I can send `PWM` signal to the motor and it responds well  I am trying to read the encoder value using `e` but it always shows `"0 0"`
I modified `motor_driver.h` according to where my encoder PINs are connected I am using Arduino mega interrupt PINS  `PIN2,PIN3` AND `PIN20,PIN21` for the left and right encoder 

```
/* *************************************************************
   Encoder driver function definitions - by James Nugen
   ************************************************************ */
   
   
#ifdef ARDUINO_ENC_COUNTER
  //below can be changed, but should be PORTD pins; 
  //otherwise additional changes in the code are required
  #define LEFT_ENC_PIN_A 20  
  #define LEFT_ENC_PIN_B 21  
  
  //below can be changed, but should be PORTC pins
  #define RIGHT_ENC_PIN_A 2 
  #define RIGHT_ENC_PIN_B 3  
#endif
   
long readEncoder(int i);
void resetEncoder(int i);
void resetEncoders();

```

MSohan14, 2024-11-20:
Hey, I've run into the same issue and am using an arduino MEGA as well.
Were you able to solve this issue?

wty-yy, 2025-01-04:
My mega 2560 can't be used either, so I wrote a tiny PID control program that supports serial port commands, it's simpler: https://github.com/wty-yy/arduino_pid_controlled_motor